Georacer/last_letter_legacy #21: Can it simulate a quadcopter?

davidbuzz, 2015-03-15:
I think it can probably be made to do it... what about u?

Georacer, 2015-03-15:
Everything is possible, but for now it is outside of my workflow. The dynamics of propellers running on static thrust and the aerodyamic drag a quad frame is not something I have studied well and will require some research.

Besides, I believe there are quite a few simulation packages for quadcopters, altough on a recent discussion with a colleague I wasn't able to give him something solid.
Are you aware of any quadcopter simulator for arduplane SITL, apart from the official one?

If there really is a lack in the market, I could try to do something in that direction.

davidbuzz, 2015-03-15:
I've not seen any open source quad simulator other than the basic python one in Ardupilot.... and it really does not do much more than fake the inputs enough to keep the SITL in the air ...

Georacer, 2015-03-15:
Hm... I was under the impression that at least the guys at ETH had cooked up something for their `autonomy` collection (mavros, mavlink, ardrone etc). If this is the case, then it might be worth investigating.
This ticket will remain open.

In the mean time, do you have a specific quad in mind? I'd need some hard data to code when I start modeling.

davidbuzz, 2015-03-15:
The ETH stuff isn't released at this time.  As to a "specific" quad... sure it's just a case of picking a well known and proven design that many Ppl have flown.  Eg one of these open designs... http://www.intorobotics.com/5-best-examples-of-how-to-build-a-diy-quadcopter/  or maybe a well know RTF like the http://3drobotics.com/iris/ ?

Georacer, 2015-03-15:
It will come down to the quad for which data is availabe. What is needed is:
- Motor specs
- Propeller specs
- All-up weight
- Accurate motor locations
- (Ideally) The matrix of inertia of a quad, but this could be estimated by virtual flight testing.

davidbuzz, 2015-03-15:
Everything except the last one is easy. Can the matrix of inertia be measured if I have the quad in my hands? I have both a home - made square one and an iris... so I can measure these ...?

Georacer, 2015-03-15:
Moderately easy. I'll point you to a couple of documents.

Georacer, 2015-06-27:
A new quadcopter model has been introduced in f2f610e1db88ea288bc9a9ecb33c4cb1fa5927d9.
I'll close this one as solved; let's discuss in new specific bug reports and questions in new issue submissions.

closed 2015-06-27 by Georacer