Georacer/last_letter_legacy #24: Vagrant configuration file for last_letter and SITL setup?

hamishwillee, 2015-03-17:
It would be useful to have a vagrant configuration file for setting up the configuration with SITL and last_letter. A good starting point would be the one [described here](http://dev.ardupilot.com/wiki/simulation-2/sitl-simulator-software-in-the-loop/setting-up-sitl-using-vagrant/).

Georacer, 2015-05-16:
I'm looking at the Vagrant option (which I've never used before) and I'm wondering if you mentioned in with the goal of automating setup for Linux, Windows or both?
From what I read at the linked wiki page, the instructions are for a Windows installation? Which one are you interested in?

Georacer, 2015-05-16:
After a couple of hours of Vagrant works, I think I'm starting to see the point. After you perform the correct installation steps for Vagrant and VBox, which are OS specific, then you can have a nice Ubuntu 14.04 environment to download and install ROS and `last_letter`.

I'm having some issues but it's a WIP, of course.

I needed to set up a VBox in this laptop to try out fresh builds anyway.

Georacer, 2015-05-16:
From what I could find, there are no Ubuntu Trusty images for VBox with a graphical environment pre-installed in the Vagrant cloud, apart from this: https://vagrantcloud.com/shadowrobot/boxes/ros-indigo-desktop-trusty64, which I have not yet tested.

Asking of the user to download the desktop is a huge overhead, so I won't go there for now.

Going without a GUI makes sense only for ardupilot SITL interfacing, whose development is stale right now. I might examine this option when the integration is in a more mature state.

Georacer, 2015-05-16:
I managed to get `last_letter` work in a full graphics environment, though. With 2 cpu cores and 4096MB of ram it was pretty sluggish, at about 1/2 to 1/3 realtime speed.

      vb.customize ["modifyvm", :id, "--memory", "4096"]
      vb.customize ["modifyvm", :id, "--ioapic", "on"]
      vb.customize ["modifyvm", :id, "--cpus", "2"]

hamishwillee, 2015-05-18:
Hi George, 

Yes, I was suggesting vagrant to ease setup on Windows/OSX.  As a rule we've assumed anyone that wants to work on Linux is smart enough to work out how to do stuff manually (of course anything that automates configuration is always welcome!).

At the moment **my** main use-case for SITL/vagrant is to provide a turnkey solution for a vehicle when testing dronekit. I really don't need any fancy graphics for that - I just tell the SITL instance to output to a second UDP port and connect that to Mission Planner to watch the action. So for my "main use case" it doesn't matter that vagrant is UI-less. 
Of course that probably missed out a heap of what you can do with last_letter. 

I understand that there has just been some success in building SITL on windows, and there are plans to integrate this with Mission Planner. Once there is an out of the box solution you might argue that getting stuff working with Vagrant is not so useful. Personally I'm still pretty happy with vagrant as I like the benefit of being able to completely clean my host whenever I like. 

Regards
Hamish