Georacer/last_letter_legacy #50: rqt_dashboard/scripts/rqt_dashboard missing?

hsu, 2015-06-22:
catkin_make install yields:

~~~
CMake Error at last_letter/rqt_dashboard/cmake_install.cmake:68 (FILE):
  file INSTALL cannot find
  "/home/hsu/projects/catkin_last_letter/src/last_letter/rqt_dashboard/scripts/rqt_dashboard".
Call Stack (most recent call first):
  cmake_install.cmake:129 (INCLUDE)

make: *** [install] Error 1
~~~

is this expected?  thanks.

Georacer, 2015-06-22:
Why did you run a `catkin_make install` instead of a simple `catkin_make`? I don't believe I have written that this command should be run, in the documentation.
There is no install file in the rqt_dashboard package, hence the error.

Georacer, 2015-06-24:
Was the problem resolved by running `catkin_make`?

Georacer, 2015-06-27:
The OP didn't provide more information on the error. This is closed, on the assumption that it is solved.

closed 2015-06-27 by Georacer